rossmann-engineering/EasyModbusTCP.NET #3: Licensing

Apollo3zehn, 2017-11-29:
I would like to give this library a try for my measurement software project. This library could help to provide data exchange via ModbusTCP. You write that it is industry approved but at the same time it is published under the Attribution-NonCommercial-NoDerivatives 4.0 International license which prohibits commercial use. 

I plan to distribute my software under the Lesser General Public License (LGPLv3) which also allows commercial use to make this software useful to more people. My software is plugin based, so I could encapsulate your library and my wrapper in a separated C# project and distribute this under your license. I do not prefer this option but I think this could be the best compromise. Or is there a chance to make this compatible to the LGPLv3 license?

Besides this: did you think about publishing your lib to NuGet? This would make development easier :)

Apollo3zehn, 2017-11-30:
I just realized that the binaries are published under a different license: Creative Commons Attribution-NoDerivatives 4.0 International Public License. Is this intended?

closed 2017-12-08 by Apollo3zehn

Gonzo4, 2017-12-19:
Hi,
I'm also confused about the licenses, can I use the binaries in a commercial project or not?
Why the different licenses?