rossmann-engineering/EasyModbusTCP.NET #5: Client responses misalignment

realselvatico, 2018-02-08:
Good morning, during some tests, I found a fairly rare situation but it can happen.
In particular, using ModBusClient, it may happen that a disalinneability occurs between the request to read some memory areas and its response.
It can happen when several readings of different areas are made, in one of these readings a communication error or other happens, in the subsequent answers the data are out of phase of a reading.

I assume there may be a problem when the stream.Read is done that the previous stream is downloaded.

I apologize in advance if I have not been clear enough, but it is a particular condition but quite annoying.

WilliamFrasson, 2018-03-24:
I see the same problem. It is quite more evident when at the end of each group of several read F03 I do a disconnect. .

WilliamFrasson, 2018-03-24:
the misalligned  can be see if one response return "BUSY" 01 83 xx xx xx , the next packet is misalligned of 5 bytes

WilliamFrasson, 2018-03-25:
this is the log from a monitor:

[25/03/2018 09:13:15] - Open port COM8 (C:\Users\frass\Dropbox\03_My_WORK\01 - SmartKeyWiFiBT\src\00_RielloDeviceTool\RielloIWH02DeviceTool\bin\Debug\RielloModbusDeviceTool.exe) 
 
[25/03/2018 09:13:15] Written data (COM8) 
    01 03 01 2c 00 6e 04 13                           ...,.n..         
[25/03/2018 09:13:15] Read data (COM8) 
    01 03 dc 30 32 34 31 33 30 38 30 32 31 30 33 38   ..Ü0241308021038 
    36 30 30 33 32 37 32 00 0c 1d b3 00 18 2c b4 55   6003272...³..,´U 
    4b 00 00 32 32 62 62 63 63 64 64 65 65 66 66 67   K..22bbccddeeffg 
    67 68 68 69 69 6c 6c ff ff ff ff ff ff ff ff ff   ghhiillÿÿÿÿÿÿÿÿÿ 
    ff ff ff ff ff ff ff ff ff ff ff 00 00 00 00 00   ÿÿÿÿÿÿÿÿÿÿÿ..... 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32   ...............2 
    31 34 33 36 35 38 37 20 20 20 20 20 20 20 20 20   1436587          
    20 20 20 00 63 00 00 00 00 00 00 00 00 00 00 00      .c........... 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1f   ................ 
    21                                                !                
[25/03/2018 09:13:15] Written data (COM8) 
    01 03 01 f4 00 03 45 c5                           ...ô..EÅ         
[25/03/2018 09:13:15] Read data (COM8) 
    01 03 06 00 00 00 00 00 00 21 75                  .........!u      
[25/03/2018 09:13:15] Written data (COM8) 
    01 03 00 c8 00 10 c5 f8                           ...È..Åø         
[25/03/2018 09:13:15] Read data (COM8) 
    01 03 20 00 00 00 00 00 00 00 00 00 00 00 00 00   .. ............. 
    00 00 00 00 00 37 32 00 0c 1d b3 00 18 2c b4 55   .....72...³..,´U 
    4b 00 00 22 5a                                    K.."Z            
[25/03/2018 09:13:16] Written data (COM8) 
    01 03 03 e8 00 05 05 b9                           ...è...¹         
[25/03/2018 09:13:16] Read data (COM8) 
    01 03 0a 07 d0 00 00 00 00 00 00 00 00 08 a9      ....Ð.........©  
[25/03/2018 09:13:16] Written data (COM8) 
    01 03 04 4c 00 19 44 e7                           ...L..Dç         
[25/03/2018 09:13:16] Read data (COM8) 
    01 03 32 00 02 00 00 00 00 02 30 02 8a 01 68 00   ..2.......0.Š.h. 
    00 00 00 00 00 00 00 00 00 00 00 00 13 00 0f 00   ................ 
    00 00 32 00 04 00 00 13 88 00 41 00 00 00 00 00   ..2.....ˆ.A..... 
    05 00 01 02 30 80 46                              ....0€F          
[25/03/2018 09:13:16] Written data (COM8) 
    01 03 04 b0 00 3a c5 0e                           ...°.:Å.         
[25/03/2018 09:13:16] Read data (COM8) 
    01 03 74 00 08 00 00 00 00 00 02 00 00 00 08 00   ..t............. 
    03 fe 4c fe 4c 00 00 01 40 00 00 00 00 00 64 00   .þLþL...@.....d. 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 00 9c da                        .......œÚ        
[25/03/2018 09:13:16] Written data (COM8) 
    01 03 05 14 00 1f 44 ca                           ......DÊ         
[25/03/2018 09:13:16] Read data (COM8) 
    01 03 3e 00 00 00 00 00 00 00 00 00 00 00 08 00   ..>............. 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 24 00 07 10 d9 00 00 00 00 00   ......$...Ù..... 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 52 dc                                          .RÜ              
[25/03/2018 09:13:17] Written data (COM8) 
    01 03 05 14 00 1f 44 ca                           ......DÊ         
[25/03/2018 09:13:17] Read data (COM8) 
    01 03 3e 00 00 00 00 00 00 00 00 00 00 00 08 00   ..>............. 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 00 00 00 00 00 24 00 07 10 d9 00 00 00 00 00   ......$...Ù..... 
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................ 
    00 52 dc                                          .RÜ              
[... 38 lines cut ...]
2018-03-25 09:13:16.131-COM8 Rx: 01 03 32 00 02 00 00 00 00 02 30 02 8A 01 68 00 00 00 00 00 00 00 00 00 00 00 00 00 13 00 0F 00 00 00 32 00 04 00 00 13 88 00 41 00 00 00 00 00 05 00 01 02 30 80 46
2018-03-25 09:13:16.139->>>> Scan: group6 - Expected : 5 secs, Elapesed Time: 6.9 secs
2018-03-25 09:13:16.151-COM8 Tx: 01 03 04 B0 00 3A C5 0E
2018-03-25 09:13:16.245-COM8 Rx: 01 03 74 00 08 00 00 00 00 00 02 00 00 00 08 00 03 FE 4C FE 4C 00 00 01 40 00 00 00 00 00 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 9C DA
2018-03-25 09:13:16.268->>>> Scan: group7 - Expected : 30 secs, Elapesed Time: 5.9 secs
2018-03-25 09:13:16.298-COM8 Tx: 01 03 05 14 00 1F 44 CA
[... cut ...]
2018-03-25 09:13:17.542-COM8 Tx: 01 03 06 40 00 20 45 4E
2018-03-25 09:13:17.603-COM8 Rx: 01 03 40 00 09 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 BF A9

as you can see the response to "01 03 05 14 00 1F 44 CA" is read badly, the first part of the response if fillexd with "00" 

I hope that you  can have time for looking this problem  because make the RTU quite instable.

thanks in advaced 
William

WilliamFrasson, 2018-03-27:
Fix the problem by disabling the receiving event and handle the recepion dldirecly into each function