rossmann-engineering/EasyModbusTCP.NET #6: VB.Net.  Trying to use EasyModbus events gives a "Ambiguous Name" compile error.

YvesLev, 2018-03-03:
For exemple, For VB, coilsChanged event is ambiguous with CoilsChanged (Case Sensitive).

So I have not been able yet to implement the coilsChanged and holdingRegistersChanged Events in VB.Net.
Always gives me compile error: 
('<elementname>' is ambiguous because multiple kinds of members with this name exist in <type> '<typename>')

But I'm quite new to this I admit, and I might not use it correctly in vb.net...

scottpage, 2018-03-23:
I just realized that I duplicated your issue with #7.  I forked this project and added "Handler" to all delegates.  That removes the ambiguity for VB.NET.

I'll push my latest branch so you can try it out from my repository.

scottpage, 2018-03-23:
See if this forked commit ed7c8c0 works for you.

YvesLev, 2018-03-24:
I downloaded this "forked" version and built it.

Yes the resultant DLL works in VB.Net now!

Thank you!

YvesLev, 2018-03-26:
However, just noticed that when I issue the StopListening() method, I receive an exception in Visual Studio.
It does not seem to stop my application though.

Here is the exception text from the Output window during debug of my application:

Exception levée : 'System.NullReferenceException' dans EasyModbus.dll
Exception levée : 'System.ObjectDisposedException' dans System.dll
System.ObjectDisposedException: Impossible d'accéder à un objet supprimé.
Nom de l'objet : 'System.Net.Sockets.Socket'.
   à System.Net.Sockets.Socket.EndAccept(IAsyncResult asyncResult)
   à System.Net.Sockets.TcpListener.EndAcceptTcpClient(IAsyncResult asyncResult)
   à EasyModbus.TCPHandler.AcceptTcpClientCallback(IAsyncResult asyncResult) dans F:\Downloads\Modbus\EasyModbusTCP_Source\EasyModbus\ModbusServer.cs:ligne 112
'Lego70909MB.exe' (CLR v4.0.30319: Lego70909MB.exe) : Chargé 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.resources\v4.0_4.0.0.0_fr_b77a5c561934e089\System.resources.dll'. Le module a été généré sans symboles.
Exception levée : 'System.InvalidOperationException' dans System.dll
System.InvalidOperationException: Pas à l'écoute. Vous devez appeler la méthode Start() avant d'appeler cette méthode.
   à System.Net.Sockets.TcpListener.BeginAcceptTcpClient(AsyncCallback callback, Object state)
   à EasyModbus.TCPHandler.AcceptTcpClientCallback(IAsyncResult asyncResult) dans F:\Downloads\Modbus\EasyModbusTCP_Source\EasyModbus\ModbusServer.cs:ligne 131
Le programme '[4268] Lego70909MB.exe' s'est arrêté avec le code 0 (0x0).