rossmann-engineering/EasyModbusTCP.NET #7: Events cannot be found using VB.NET

scottpage, 2018-03-21:
The naming conventions used for the ModbusClient and ModbusServer work as intended in C#, because C# is case sensitive.  The events are useless for VB.NET development because it is not case sensitive.

The following is an example taken from the source.

<code>
public delegate void NumberOfConnectedClientsChanged();
public event NumberOfConnectedClientsChanged numberOfConnectedClientsChanged;
</code>

Please update your naming convention.  Instead of just altering the case, please use a different name for the event and the delegate.  Preferably add the word "Handler" to the end of a delegate and leave the Event named the way it is.

scottpage, 2018-03-23:
Duplicate of #6

rossmann-engineering, 2018-03-27:
Yes you are probably right and I updated that. I was aware of that problem, but I didn't want to loose compatibility to previous versions.
But there will be some major changes anyway.
Thanks for your contribution

closed 2018-03-27 by rossmann-engineering