rossmann-engineering/EasyModbusTCP.NET #8: License question

dazzalogan, 2018-06-20:
Hi there,

Apologies in advance, but I don't understand the licensing of EasyModbus.

I'd like to add a couple of minor properties (serialport handshake control lines) to EasyModbus and use it in my own software. However, I don't understand the license implications of doing this.

Am I allowed?
Do I have to offer the source of my entire software if I use EasyModbus?

Thank you

Regards,
Darren

rossmann-engineering, 2018-07-02:
See front page

closed 2018-07-02 by rossmann-engineering

dazzalogan, 2018-07-03:
Hi,
OK - so it cannot be used unless full source code of the entire project is made public?
Is there any way we can get around this? for a fee, perhaps?

Thank you

rossmann-engineering, 2018-07-08:
Hi,

as far as I know you don't have to make your whole source code public, you only have to point our your customers that you used the library, and give them also access to this repository. 

Otherwise there would be an option for a commercialized version.

dazzalogan, 2018-07-09:
Hi there,

The license is full of "legal jargon" which is difficult to understand (in my opinion).
Some parts "sound like" you have to offer source of the full project, other parts don't. It's confusing.

I suppose what I should ask you is; when you chose a license model, what did you actually intend?

There is a "lesser" license which is a less restrictive, or of course you could have offered it completely free :-)