rossmann-engineering/EasyModbusTCP.NET #11: License GPL or LPGP

Gonzo4, 2018-07-21:
Hi,

sorry for opening the next licensing issue, but is the project licensed under the GPL (see startpage) or under the LGPL, as you assumed in @dazzalogan issue #8: 
> as far as I know you don't have to make your whole source code public, you only have to point our your customers that you used the library, and give them also access to this repository.

And please add the license to this repository (LICENSE -file) to make it clear ;)
My project is licensed under the MIT license and I have to change to another library if this library is licensed under the GPL, so please change the license to the LGPL version ;)

dazzalogan, 2018-09-05:
Hi,

I agree. It would make the licensing a lot easier if it was reduced to LGPL!

Gonzo4, 2020-09-25:
I see its now licensed under the MIT license, thanks 👍

closed 2020-09-25 by Gonzo4