rossmann-engineering/EasyModbusTCP.NET #16: Connected is still true after disconnection

Mark-81, 2018-10-29:
In a C# application I use EasyModbus to connect to a remote server:

```
ModbusClient client = new ModbusClient()
{
    IPAddress = "192.168.1.100",
    Port = 502,
    ConnectionTimeout = 200
}

try
{
    client.Connect();
}
catch (EasyModbus.Exceptions.ConnectionException)
{
    throw;
}
catch (System.Net.Sockets.SocketException)
{
    throw;
}
```
then in other parts of my application I try to read/write only if the client is still connected:

```
if (client.Connected)
{
    try
    {
        int[] readHoldingRegisters = client.ReadHoldingRegisters(1000, 10);
        // do something
    }
    catch (Exception)
    {
        throw;
    }
}
```
If while the application is running I disconnect my laptop from the WiFi network, the Connected property is still true even after a long time. I get a "out of range" exception when I try to read the registers, but I wonder why that property doesn't turn to false immediately. At least, I would expect a "timeout" exception when I try to read/write something.

Is there a better approach to know the server is not reachable anymore?

rossmann-engineering, 2018-12-27:
Yes thats an issue which can't be 100% solved. If the connection breaks up from server side (without it has been properly closed) there is no way to detect that. 
Possibly the Method "Available" could be helpful?

closed 2018-12-27 by rossmann-engineering

enricobenedos, 2020-12-14:
I've the same problem and at this point the `Connect` property is very misleading/useless. A good idea can be to remove it or try to fix the problem.
This is the generated exception (unfortunately in italian):
```
System.IO.IOException: Impossibile leggere dati dalla connessione del trasporto: Impossibile stabilire la connessione. Risposta non corretta della parte connessa dopo l'intervallo di tempo oppure mancata risposta dall'host collegato. ---> System.Net.Sockets.SocketException: Impossibile stabilire la connessione. Risposta non corretta della parte connessa dopo l'intervallo di tempo oppure mancata risposta dall'host collegato
   in System.Net.Sockets.Socket.Receive(Byte[] buffer, Int32 offset, Int32 size, SocketFlags socketFlags)
   in System.Net.Sockets.NetworkStream.Read(Byte[] buffer, Int32 offset, Int32 size)
   --- Fine della traccia dello stack dell'eccezione interna ---
   in System.Net.Sockets.NetworkStream.Read(Byte[] buffer, Int32 offset, Int32 size)
   in EasyModbus.ModbusClient.ReadHoldingRegisters(Int32 startingAddress, Int32 quantity)
   in (<StatusUpdateCycle>d__28& )
```
A good idea can be to catch specifically the `SocketException` and force the connection close also setting up `Connect` property to false.