rossmann-engineering/EasyModbusTCP.NET #20: Recursive calls for retries

mcurry-va, 2019-02-15:
Very bad practise to use recursion when other methods are available.

I came across an issue whereby framing errors were occuring on the bus and the code-route was zeroing the retry count sometimes through, causing a stack overflow.

I would recommend a complete refactor to avoid recursion.

rossmann-engineering, 2019-02-15:
I am always open to suggestions, but only complaining about the code quality unfortunately doesn’t help.

closed 2019-02-15 by rossmann-engineering

mcurry-va, 2019-02-19:
This is an actual bug report though!
The cause was from framing errors being caught on the serial port, for which the windows driver inserts '?'
The stack _should_ retry 3 times and then throw a CRC error, however a certain code route zeros the retry count when the slave address in the reply does not match.

My original report was that the fundamental recursive calls need to be fixed rather than this specific bug.