rossmann-engineering/EasyModbusTCP.NET #22: Test and user , get some error .

YuHanChen04, 2019-02-22:
Sorry for my English.
-----------------------------------------
I followed the video of  [David Kaleta](https://www.youtube.com/watch?v=OPAIFIFWFA4) on Youtube , my code and tool as same as his , and I got some error  :
EasyModbus.Exceptions.StartingAddressInvalidException:
 'Starting address invalid or starting address + quantity invalid'

Here is my code : 
```
using System;
using System.Collections.Generic;
using System.ComponentModel;
using System.Data;
using System.Drawing;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using System.Windows.Forms;
using EasyModbus;

namespace Test_modbus
{
    public partial class tb_stat : Form
    {
        ModbusClient modbusClient;
        public tb_stat()
        {
            InitializeComponent();
        }

        private void bt_Connect_Click(object sender, EventArgs e)
        {
            try
            {
                modbusClient = new ModbusClient(tb_Modbus_Server_IP.Text, 502);    //Ip-Address and Port of Modbus-TCP-Server
                modbusClient.Connect();
                lb_stastus.Text = "Connected";
                tmr_Moudus_Com.Enabled = true; 
            }
            catch (Exception ex)
            {
                lb_stastus.Text = ex.ToString();
                throw;
            }            
        }
        private void lb_stat_Click(object sender, EventArgs e)
        {
        }
        private void bt_Disconnect_Click(object sender, EventArgs e)
        {
            modbusClient.Disconnect();
            lb_stastus.Text = "Disconnect.";
            tmr_Moudus_Com.Enabled = false;
        }

        private void timer1_Tick(object sender, EventArgs e)
        {
            tmr_Moudus_Com.Enabled = false;
            
            modbusClient.WriteMultipleCoils(4, new bool[] { true, true, true, true, true, true, true, true, true, true });    //Write Coils starting with Address 5
            bool[] readCoils = modbusClient.ReadCoils(0, 10);                        //Read 10 Coils from Server, starting with address 10
            int[] readHoldingRegisters = modbusClient.ReadHoldingRegisters(0, 10);    //Read 10 Holding Registers from Server, starting with Address 1

            aGauge1.Value = readHoldingRegisters[0];
            tmr_Moudus_Com.Enabled = true;
        }

        private void trackBar1_Scroll_Scroll(object sender, EventArgs e)
        {
            aGauge1.Value = trackBar1.Value;

        }

        private void aGauge1_ValueInRangeChanged(object sender, AGaugeApp.AGauge.ValueInRangeChangedEventArgs e)
        {

        }
    }
}
 
```

Please help .
Thank you so much ..

Padanian, 2019-02-22:
Node Address can't be 0. 
0 is reserved for broadcast messages (write only)

rossmann-engineering, 2019-02-27:
Hi,

your Server responds with an error code. One reason could be that one of the functioncodes are not supported by your server, or your trying to read a address which does not exist.

So it is an issue related with your server, not with your code.

@Padanian: Thanks a lot for your contribution, but I don't think thats the reason, because for Modbus TCP the Slave ID isn't used (only for Gateways) and it has to be set to 0xFF or 0.

closed 2019-02-27 by rossmann-engineering

Padanian, 2019-02-27:
Sorry, I didn't realize it was modbusTCP.