rossmann-engineering/EasyModbusTCP.NET #24: Unable to write data to the transport connection in Timr

Styxer, 2019-02-26:
I'm working with BNR X20CP0410 plc
which I'm reading 2 discrete input from address 1

when the timer interval is set 1 millisecond all is working 

but when i bump the interval to 1000 milliseconds and i put a breakpoint on one of the lines at the timer i get the error "Unable to write data to the transport connection: An established connection was aborted by the software in your host machine"

and when i event bump the interval even more to 5000 milliseconds i got the error on the first entry to the timer with or without breakpoint

 [ private void initModBusClient()
        {       
            client = new ModbusClient()
            {
               IPAddress = "192.168.66.215",
               Port =  502,
               ConnectionTimeout = 5500
            };
            client.Connect();
            client.ConnectedChanged += client_ConnectedChanged;
        }

 void client_ConnectedChanged(object sender)
        {
            throw new NotImplementedException();
        }

  private void timer1_Tick(object sender, EventArgs e)
        {

            var resDes = client.ReadDiscreteInputs(1, 1);

            System.Diagnostics.Trace.WriteLine(resDes[0]);

           
        }](url)

Padanian, 2019-02-26:
Pretty normal. Your PLC is not design to wait seconds for the connection. It times out and drops.

rossmann-engineering, 2019-02-27:
Yes, I agrre with Padanian's comment, I think your Server times out in disconnects. The best way to handle that, is to enclose the request in a Try-catch block and disconnect (and then reconnect) if the exception is thrown.

The "connected" property infortunately does not work reliably in that case, because it does not detect if the connection was meanwhile closed from Server-side

Styxer, 2019-02-27:
you are both right, it turns out the the PLC timeout was 3 seconds, if i changed it to 6 seconds and my timer to 5 seconds as i wanted all is working

closed 2019-02-27 by Styxer