rossmann-engineering/EasyModbusTCP.NET #26: best way to handle this error: Unable to write data to the transport connection: An existing connection was forcibly closed by the remote host.

aventurella, 2019-05-04:
I frequently see this error when running either the example GUI apps in this repo or from my own application.

Currently, when I get it, I cleanup my client and reconnect. But I'm curious if this is the best way to handle this.

bergi9, 2019-08-01:
Same applies to me. I even check `client.Connected` before I use `client.WriteMultipleCoils(0, values);` and the error still appears.

aventurella, 2019-08-15:
I also get this occasionally:

```
A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.
```

My device is connected directly to the machine I'm reading it from so it's odd that it's running into this issue.