rossmann-engineering/EasyModbusTCP.NET #32: EasyModbus TCP slows down the windows application

nayanavv, 2019-10-14:
Hi,
I'm using EasyModbus TCP to connect to a slave device from a windows application (WinForms).
I'm sending and receiving float values using the methods inside the timer for Modbus communication since the values should be communicated continuously.

modbusClient.WriteMultipleRegisters(register, ModbusClient.ConvertFloatToRegisters((float)variable, ModbusClient.RegisterOrder.HighLow)); (for writing to Slave)

and

ModbusClient.ConvertRegistersToFloat(modbusClient.ReadHoldingRegisters(register, 2), ModbusClient.RegisterOrder.HighLow);
(for Reading from Slave)

It works well when the number of methods used inside the timer is less,
but the application slows down when the number of methods increases. 

Is there any way to speed up the whole communication?

Thanks 
Nayana

Padanian, 2019-10-14:
I don't have any problem with this and don't see any speed
reduction.Can you please share your code or would you please explain
how you call the methods and how often?

----- Messaggio Originale -----
Da: "rossmann-engineering/EasyModbusTCP.NET" 
A:"rossmann-engineering/EasyModbusTCP.NET" 
Cc:"Subscribed" 
Inviato:Mon, 14 Oct 2019 10:49:53 +0000 (UTC)
Oggetto:[rossmann-engineering/EasyModbusTCP.NET] EasyModbus TCP slows
down the windows application (#32)

	Hi,
 I'm using EasyModbus TCP to connect to a slave device from a windows
application (WinForms).
 I'm sending and receiving float values using the methods inside the
timer for Modbus communication since the values should be communicated
continuously. 

	modbusClient.WriteMultipleRegisters(register,
ModbusClient.ConvertFloatToRegisters((float)variable,
ModbusClient.RegisterOrder.HighLow)); (for writing to Slave) 

	and 

	ModbusClient.ConvertRegistersToFloat(modbusClient.ReadHoldingRegisters(register,
2), ModbusClient.RegisterOrder.HighLow);
 (for Reading from Slave) 

	It works well when the number of methods used inside the timer is
less,
 but the application slows down when the number of methods increases. 

	Is there any way to speed up the whole communication? 

	Thanks
 Nayana 

	—
"https://github.com/rossmann-engineering/EasyModbusTCP.NET/issues/32?email_source=notificationsu0026email_token=AB3I2BFLFSVQ2Z4UUT4JK7LQORFEVA5CNFSM4JANPYH2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HRRYJLQ",
"url":
"https://github.com/rossmann-engineering/EasyModbusTCP.NET/issues/32?email_source=notificationsu0026email_token=AB3I2BFLFSVQ2Z4UUT4JK7LQORFEVA5CNFSM4JANPYH2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HRRYJLQ",
"name": "View Issue" }, "description": "View this Issue on GitHub",
"publisher": { "@type": "Organization", "name": "GitHub", "url":
"https://github.com" } } ]

Links:
------
[1]
https://github.com/rossmann-engineering/EasyModbusTCP.NET/issues/32?email_source=notifications&amp;email_token=AB3I2BFLFSVQ2Z4UUT4JK7LQORFEVA5CNFSM4JANPYH2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HRRYJLQ
[2]

nayanavv, 2019-10-21:
Hi Padanian,

I have resolved this issue by multithreading. 
The slow Modbus read and write applications were making the whole application not responsive(slow).

closed 2019-10-21 by nayanavv