rossmann-engineering/EasyModbusTCP.NET #33: How to use the swap functionality

JawadJaber, 2019-10-15:
I tried to use this library with some devices, and it did not succeed. although it is performing well with many devices..

I can attribute this behavior to CRC and swapped registers, but I don't know how to apply it..

I found that some functions have no use in the library such as this:

` public static float ConvertRegistersToFloat(int[] registers, RegisterOrder registerOrder)
        {
            int [] swappedRegisters = {registers[0],registers[1]};
            if (registerOrder == RegisterOrder.HighLow) 
                swappedRegisters = new int[] {registers[1],registers[0]};
            return ConvertRegistersToFloat(swappedRegisters);
        }`

How to use this for swapped configuration, it will be very helpful if it is included in the UI.

rossmann-engineering, 2019-10-31:
I am not sure if I understood the problem. The CRC is created after the request to the Slave has been creasted and before it is send. There is no way and no need to manipulate.

JawadJaber, 2019-10-31:
Thanks you..
In which case we can use above mentioned method? How to know if CRC is HighLow or LowHigh?

rossmann-engineering, 2019-10-31:
The Order of the CRC is specified in the Modbus spec. There is no need to change. 

The above functions are used if you transfer a "float" value using Modbus. Usually a float value is 32 bits long while a Register is 16 bits long. The float value then has to be separated into two Registers, and Merged afterwards.

closed 2019-11-17 by JawadJaber