rossmann-engineering/EasyModbusTCP.NET #38: Cannot set Parity to None

darkmansad, 2020-01-21:
Cannot set the parity to 'None' . 
If we make the following code: 

var _modbusClient=new ModbusClient();
_modbusClient.Port = 6;
_modbusClient.Parity = System.IO.Ports.Parity.None;
_modbusClient.StopBits = System.IO.Ports.StopBits.Two;
_modbusClient.Connect(); 

this code will attempt to connect with parity Even, i think this is a problem of the setter where the parity cannot be set if the 'if (serialport != null)'
If the constructor is adapter and is passed also the 'COM6' parameter the setter for  paritity parameter will work.

Padanian, 2020-06-25:
Isn't it Port the TCP/IP port? Shouldn't it be Serialport?

darkmansad, 2020-06-25:
> Isn't it Port the TCP/IP port? Shouldn't it be Serialport?

It's for serial port.

Padanian, 2020-06-25:
No, it isn't.
[image]

Padanian, 2020-06-25:
Your example works if amended like this:

`if !_databaseTypeIsTCP{ 
//set serial data 
_modbusClient.SerialPort = "COM6"

closed 2020-06-25 by darkmansad