rossmann-engineering/EasyModbusTCP.NET #43: No support for .NET standard

1806599krottje, 2020-06-05:
Hi,
I would like to use your NuGet-package in .NET Standard, but that's unsupported at the moment.
Is there a possibility or near future planning on support for .NET Standard?
Thanks in advance!

1806599krottje, 2020-06-05:
If I try to run the framework in Xamarin for example, I get this error:
`System.TypeLoadException: 'Could not load type of field 'EasyModbus.ModbusClient:serialport' (19) due to: Could not resolve type with token 01000010 from typeref (expected class 'System.IO.Ports.SerialPort' in assembly 'System, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089') assembly:System, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 type:System.IO.Ports.SerialPort member:(null)'`

easis-enttia, 2020-06-18:
I had the same issue and fixed it by downloading the NuGet-package called System.IO.Ports.

Padanian, 2020-07-16:
I tentatively submitted the repository ported to .NET5 
Wishing Rossmann merge the pull (and any other pending pull requests), or anyone interested forks my repo

closed 2020-10-25 by bjarnekrottje