rossmann-engineering/EasyModbusTCP.NET #44: UnitIdentifier & Modbus TCP

pechezerov, 2020-06-11:
I use ModbusServer like this:
```
public SlaveWrapper(int port)
{
      server = new EasyModbus.ModbusServer();
      server.Port = port;
      server.SerialFlag = false;
      server.UDPFlag = false;
      server.LogDataChanged += someMethod;
      server.Listen();
}
```

If I connect to the server with SlaveId = 1:
[image]

If I connect to the server with SlaveId = 2 (or any other value):
[image]

In this [page](http://easymodbustcp.net/en/net-modbusserver-properties)
I see:

> public byte UnitIdentifier
> gets or sets the unit identifier (Slave-ID). Normally only nessesary for Modbus-RTU (Standard is 1)

But there is slightly different approach from modbus.org [see page.23](http://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf)
> On TCP/IP, the MODBUS server is addressed using its IP address; therefore, the
> **MODBUS Unit Identifier is useless**. The value 0xFF has to be used. 

This approach is followed, for example, by [libmodbus](https://github.com/stephane/libmodbus/).
Default UnitId for all Modbus TCP requests is 255.

What do you think, does this mean that a UnitIdentifier match check should not be performed for the TCP server?

Padanian, 2020-06-11:
Exactly. The node address in modbusTCP is the TCPIP address.
There is an exception for modbusRTU over TCP. In this case the TCP address is the gateway's, which shall route the request to the node address specified in the unitID.
But modbusRTU over TCP was never officially approved, and Easymodbus does not cover it.

pechezerov, 2020-06-11:
So, I have to use UnitId = 1 anyway?

Padanian, 2020-06-11:
Strictly speaking, it shall make no difference whatsoever. 
If it abides strictly to specs, you set it to 0xFF and the server shall ignore it.

closed 2020-06-12 by pechezerov