rossmann-engineering/EasyModbusTCP.NET #49: When the TCP server is disconnected

NowUp, 2020-09-30:
Hello, In the process of using, I found a problem, this problem seriously affected the use.

In the state of connection, when the server TCP terminates the service, no abnormal feedback is received in any read and write operation of the client, the value of Connected property is true, and after the server service is restarted, it still cannot operate normally.

I used Modbus Slave for TCP communication simulation.

BatangTurun, 2020-10-18:
I had the same concern, what to do if the connection (LAN) went down.

My solution was to ping the TCP device every 500mS to test for connection and take appropriate action if LAN went down.

Padanian, 2020-10-18:
Same here. I ping it every 1000ms to keep the socket open

rossmann-engineering, 2020-10-19:
Yes that is indeed in issue, which can't be reliably solved (At least I don't know how). That happens if the connection breaks up, without properly closing it. 
Any request should run into a timeout. It is usually the way to reconnect if any exception is thrown. Also there is the "Available" Method which pings the server to check for its availability.