rossmann-engineering/EasyModbusTCP.NET #52: Exception: Serial Port already opened

hannekedo, 2020-11-23:
I use the EasyModbus lib for communication with several modbus devices over RS485.

The following happens quite regularly:

- Modbus communication is up and running
- All of a sudden the following exception is thrown:
`ModbusMainAction EasyModbus.Exceptions.SerialPortNotOpenedException: serial port not opened`
- It tries to reconnect, but then an Exception is thrown with the following message:
`Serial Port already opened`
- When this happens the modbus client is disconnected and an attempt is done to reconnect, but upon reconnecting I again get `Serial Port already opened`, so this is an endless loop until I restart the application.

I use version 4.4.1.0 of EasyModbus, which I realize is old. However I tried updating and that also doesn't seem to work flawlessly. Therefore my question is if this is a known issue to this version? Is there something that can be done while using 4.4.1.0 that could resolve this? Do you have any clue as to why this could happen? Thanks for your help!

Padanian, 2020-11-23:
Never had this problem whatsoever before. I've been using EasyModbus for quite some time, with no such issue. I'm afraid this could be caused by a faulty serial converter. Possibly a USB to RS485 converter going into sleep mode unexpectedly.

martin1ehm, 2020-11-23:
@hannekedo I also never had this problem but I strongly recommend to use an USB converter with FTDI chip and disable the power savings settings for USB devices. https://www.windowscentral.com/how-prevent-windows-10-turning-usb-devices
[image]

hannekedo, 2020-11-24:
Thanks for the quick replies and advice. I am going to look into the cables used. I am not using serial to USB though, but various different cable setups to different devices. It happens in all of these setups, which made me think it could be a library issue. 

Forgot to mention: the OS it runs on is Ubuntu 18.04.