rossmann-engineering/EasyModbusTCP.NET #57: Performance issues on Linux / NET5 because of System.IO.Ports

neographikal, 2021-02-02:
As reported over here: https://github.com/dotnet/runtime/issues/2379

When using on Linux / .NET core 5 (not formally supported yet, still it works) the cpu usage of Modbus RTU is terrible, barely usable. I'm talking half a core to do 3 requests/sec to some slave devices. The implementation of System.IO.Ports is not good, which is not a problem on WIndows (no cpu usage at all), but on Linux it goes through the roof.

Testing / using it: https://github.com/neographikal/Modbus2Mqtt

neographikal, 2021-02-10:
I switched out the default implementation with the Mono implementation, but on Windows this resulted in non functional serial communication.

elclaudio, 2021-07-06:
hi
Could you explain how to set it up to compile and run the examples on http://easymodbustcp.net/codesampleseasymodbustcp-net ?
I'm trying to compile and run them on a raspberry pi 4 which should poll a modbus slave. I could not choose net5 and if I select framework 4, it compile, but when I run it under linux it crash:
`Unhandled Exception:
System.TimeoutException: No Response from Modbus Slave
  at EasyModbus.ModbusClient.ReadHoldingRegisters (System.Int32 startingAddress, System.Int32 quantity) [0x00817] in <8140caba19fe49389617683c5fa07156>:0 
  at EasyModbus.ModbusClient.ReadHoldingRegisters (System.Int32 startingAddress, System.Int32 quantity) [0x00831] in <8140caba19fe49389617683c5fa07156>:0 
  at EasyModbus.ModbusClient.ReadHoldingRegisters (System.Int32 startingAddress, System.Int32 quantity) [0x00831] in <8140caba19fe49389617683c5fa07156>:0 
  at EasyModbus.ModbusClient.ReadHoldingRegisters (System.Int32 startingAddress, System.Int32 quantity) [0x00831] in <8140caba19fe49389617683c5fa07156>:0 
  at modbustest3.Program.Main (System.String[] args) [0x00043] in <28b1a9dd07c24554a2210d774c0a8882>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.TimeoutException: No Response from Modbus Slave
  at EasyModbus.ModbusClient.ReadHoldingRegisters (System.Int32 startingAddress, System.Int32 quantity) [0x00817] in <8140caba19fe49389617683c5fa07156>:0 
  at EasyModbus.ModbusClient.ReadHoldingRegisters (System.Int32 startingAddress, System.Int32 quantity) [0x00831] in <8140caba19fe49389617683c5fa07156>:0 
  at EasyModbus.ModbusClient.ReadHoldingRegisters (System.Int32 startingAddress, System.Int32 quantity) [0x00831] in <8140caba19fe49389617683c5fa07156>:0 
  at EasyModbus.ModbusClient.ReadHoldingRegisters (System.Int32 startingAddress, System.Int32 quantity) [0x00831] in <8140caba19fe49389617683c5fa07156>:0 
  at modbustest3.Program.Main (System.String[] args) [0x00043] in <28b1a9dd07c24554a2210d774c0a8882>:0`

The test app work under windows just fine, but I need it to run under linux !