rossmann-engineering/EasyModbusTCP.NET #59: Modbus RTU per Modbus TCP Adapter: Slave not connected

Nick135, 2021-02-17:
I use the lib to communicate with a Modbus RTU Device over an Modbus TCP to RTU Gateway.

When the Gateway is connected, but not the device (UnitIdentifier), I get no error on modbusClient.Connect().

To test the connection I make a test read, but also no exception.
I see in the code, that there is no error handling for this return message data[8]==0x0B.

if (data[7] == 0x83 & data[8] == 0x0B) //missing

Is there any description what 0x0B means?

Otherwise I add 'throw new EasyModbus.Exceptions.ConnectionException'

waedg, 2021-07-07:
Hi @Nick135, how you can communicate with Modbus RTU over a Modbus TCP. In the documentation it is mentionned :
"byte UnitIdentifier
Gets or Sets the Unit identifier in case of serial connection (Default = 0)"
I think a communication with Geteway is not supported by this DLL

Padanian, 2021-07-07:
I was able to do it through an adfweb rtu to tcp converter.
There's just one minor thing not working, which is the broadcast message, because the gateway is unable to properly handle unit identifier 0, which is a reserved address in tcp, but a broadcast address on rtu.

Nick135, 2021-07-08:
I use the Library with a "Serial Device Server".
With the modification the lib works fine since month. But I always use the same Device Server with one or two devices.

````
var modbusClient = new ModbusClient(ip, port)
{
    UnitIdentifier = slaveId,
    ConnectionTimeout = ConnectionTimeOut,
};