rossmann-engineering/EasyModbusTCP.NET #60: Multiple RTU connection to clients

NintendoPlayer, 2021-04-22:
Hi!

I don't really know if it's possible, but I would like to ask if, right now, I can connect to more than one modbus rtu client.  I want to know if my app can read addresses from two different clients on the same COM port and with the same BaudRate.

Would creating two different ModbusClient instances work?

Thank you!

Padanian, 2021-04-22:
If by client you mean Master, in any modbusRTU network there can be only one Modbus Master and as many as 249 Modbus Slaves, provided that recent politically-correct lunacy requires the word "Master" to be replaced by "Primary" and "Slave" to be replaced by "Secondary"

NintendoPlayer, 2021-04-22:
Yes, I'm also very confused by all this word-change madness.  I understand it, but I think it's confusing.  Anyway, I was refering to connect to more than one slave.

Padanian, 2021-04-22:
Of course you can connect up to 249 slaves to one master, provided each of them have unique slave address.