rossmann-engineering/EasyModbusTCP.NET #61: Modbus RTU Unit Identifier 128 and above always sending 63!

NGame1, 2021-05-20:
Hi everyone, 
I'm working with EasyModbus TCP on my .NET 5 project that running on Armbian.
A strange issue that I found is that when you specify a Unit Identifier above 128 (128 itself or greater) in the serial port the sent data will be like this.
63,6,0,0,0,1,86,27
witch means unit identifier is 63.
My code is: 
```
            var client = new ModbusClient("COM4") { Baudrate = 19200, StopBits = StopBits.One, Parity = Parity.None };
            while (true)
            {
                try
                {
                    client.UnitIdentifier = 128;
                    client.Connect();
                    client.WriteSingleRegister(0, 1);
                    //Console.WriteLine(client.sendData[0]);
                    Thread.Sleep(100);
                }
                catch (Exception)
                {
                }
            }
```

Padanian, 2021-05-20:
128 and above require 8 bit data.

Try add databits = 8

Padanian, 2021-05-20:
> You mean client.DataBits = 8?
> Databits property is not present in ModbusClient.

You can change the databits property in the COM4 serialport properties.

Padanian, 2021-05-20:
Did it work?