rossmann-engineering/EasyModbusTCP.NET #68: Type of EasyModbus.ModbusServer.HoldingRegisters.localArray is not suitable

aigefjk, 2021-09-09:
public short[] localArray defined in EasyModbus.ModbusServer.HoldingRegisters, which is not suitable defined as
a short[] type because sometimes we need to write a value bigger than 32767 (UInt16) to a single register. I suggest we could change short[] to int[] to solve this issue.

BTW,  type of EasyModbus.ModbusServer.InputRegisters.localArray has the same issue!
[image]

Padanian, 2021-09-09:
You shall use ushort, not int.
Int is signed 32bit, while ushort is unsigned 16bit

aigefjk, 2021-09-09:
> You shall use ushort, not int.
> Int is signed 32bit, while ushort is unsigned 16bit
Maybe you don't get what problem I met.
I mean if Modbus server itself wants to update the holding registers by set its localArray,  localArray whose type is short[] can not receive value greater than 32767 ,  value of ushort type also has the same problem.