rossmann-engineering/EasyModbusTCP.NET #69: There are differences between index  of modbus server register array  and index of modbus client register array

aigefjk, 2021-09-09:
When I test V5.5 library，I found there are differences  between Modbus server register index  and Modbus client register index.

Index of  registers Modbus client class used is from 0 , while index of  registers Modbus server class used is from 1.

e.g:

If you want to get value in holding register 0,   you need pass index 0 to modbus client method but in modbus server method you must pass index 1 to the method.

 public void WriteSingleRegister(int Addr, short Value)
        {
            MyServer.holdingRegisters.localArray[Addr + 1] = Value;
        }

Padanian, 2021-09-09:
For historical reasons, user reference numbers were expressed as decimal numbers with a starting offset of 1. However MODBUS uses the more natural software interpretation of an unsigned integer index starting at zero.

So a MODBUS message requesting the read of a register at offset 0 would return the value known to the application programmer as found in register 4:00001 (memory type 4 = output register, reference 00001).

aigefjk, 2021-09-09:
> For historical reasons, user reference numbers were expressed as decimal numbers with a starting offset of 1. However MODBUS uses the more natural software interpretation of an unsigned integer [...]

---------------------------------------------------------------------------
Modbus client myClient.WriteSingleRegister(0, Value) update value in register 4:00001, but in modbus server  localArray[1] presents 4:00001 , that not keep the same style.

xsvold, 2024-08-21:
I can confirm this.

I can read and write using the same address on the server side. I can also due the same on the client side. 

However, using another client, shows, that the data gets writen to different locations. (Client is +1)