rossmann-engineering/EasyModbusTCP.NET #74: When the server is actively disconnected(当服务器端主动断开时)

jkh404, 2022-01-25:
我用的EasyModbusTCP.NET v5.6
当TCP服务器断开连接时，客户端并不能获取到反馈，依然能够读取和写入(显然这是错误的)。
不过，我查看EasyModbusTCP.NET 项目的 ModbusClient.cs 源代码后， 将Connected的代码改为以下代码即可得到正确的反馈。

``` c#
public bool Connected
{
	get{
                if (serialport != null)
                {
                    return (serialport.IsOpen);
                }

                if (udpFlag & tcpClient != null)
                    return true;
                if (tcpClient == null)
                    return false;
                else
                {
                    return connected && tcpClient.Connected;

                }

	}
}
```

[image]
也许这不是最好的处理方法，但经过我的测试确实解决了这个问题。哈哈😁
en:
I use EasyModbusTCP.NET v5.6
When the TCP server is disconnected, the client cannot get feedback and can still read and write (obviously wrong). 
However, after I looked at the ModbusClient.cs source code of the EasyModbusTCP.NET project, I changed the Connected code to the following code to get the correct feedback.
Maybe this is not the best way to deal with it, but my tests did solve the problem.😁

jkh404, 2022-01-25:
这是最终效果
[image]

wmx1033, 2024-03-07:
的确实现了，顶一个