rossmann-engineering/EasyModbusTCP.NET #77: Network Stream

harjitsinghhpk, 2022-01-25:
If connected with TCP/IP, the connection aborted if there is no poll for sometime with server. It end up with below exception:

System.Net.Sockets.SocketException : Unable to read data from the transport connection: An established connection was aborted by the software in your host machine.

Is there anyway to keep the connection live even there is no poll until disconnected from slave.

Padanian, 2022-01-25:
You can poll any register every n seconds, where n < disconnection timeout
Or you can check if connected before attempting reading or writing.

Padanian, 2022-01-25:
Polling a register is not a hack. 
Anyway, the right thing to do is to set the tcp timeout on your server to a sensible value, and/or prevent the timeout to expire.

harjitsinghhpk, 2022-01-25:
Or
Can we add a system.timer in the master library, which will keep the connection live by writing and reading on stream after 10 seconds. 
The write and read can  be blocked if there is any request on the function codes. If there is not request received to the library for 10 seconds it will use the above steam read/write to keep the connection alive.

harjitsinghhpk, 2022-01-25:
> Polling a register is not a hack. Anyway, the right thing to do is to set the tcp timeout on your server to a sensible value, and/or prevent the timeout to expire.

: Issue with polling register is, user have to define the register which is available in the slave to poll always.

Padanian, 2022-01-25:
Well, if you plan to connect via modbus to a slave, you should at least be aware of one or more register addresses actually available on that slave. Don't you think?

Padanian, 2022-01-25:
> Or
> Can we add a system.timer in the master library, which will keep the connection live by writing and reading on stream after 10 seconds.
> The write and read can be blocked if there is any request [...]

Terrible idea in the class definition. You can do it though in your implementation.

harjitsinghhpk, 2022-01-26:
> Well, if you plan to connect via modbus to a slave, you should at least be aware of one or more register addresses actually available on that slave. Don't you think?

Hi,
Definitely there will be some registers in the slave, but it will be uncertain extra work on the master interface who so ever is implementing the library in their project. In case where a user just want to get the data when a request is sent. 

To elaborate further, say i have prepared a modbus master tool using the library. I can connect it to multiple slave for testing/data collection. Each slave may not have the same address as other, in this case a additional lookup for defining the keep alive registers need to be taken care. I was just looking a way out to avoid this.

Padanian, 2022-01-26:
The issue here, IMHO, is that you have a timing-out *slave problem*, that you'd like to solve on the master.
If so, why don't you just close the connection after polling and reopen the connection before polling, so that the slave will never time out.

harjitsinghhpk, 2022-01-26:
> The issue here, IMHO, is that you have a timing-out *slave problem*, that you'd like to solve on the master.
> 
> If so, why don't you just close the connection after polling and reopen the connection before polling, so that the slave will never time out.

Probably the best to do i guess. 👌🏻