rossmann-engineering/EasyModbusTCP.NET #84: ModbusClient.Available() Has memory leak

3D-Lasers-Lab, 2022-08-04:
Every time the Available() method is called the program uses about another 100k or RAM.

PRIMETSS, 2022-12-05:
Here is the associated code for `ModbusClient.Available()` 

Doesn't look like either implement IDisposable `System.Net.NetworkInformation.Ping` so doubt `using(System.Net.NetworkInformation.Ping pingSender = new System.Net.NetworkInformation.Ping())` would help?
There are some cleanup Internal Dispose methods but they are not public.

Are you sure it's a memory leak and not  just a delay until the Garbage Collector cleans it's up once goes out of scope?
100K does seem high, how did you check for that use?

`public` bool Available(int timeout)
        {
            // Ping's the local machine.
            System.Net.NetworkInformation.Ping pingSender = new System.Net.NetworkInformation.Ping();
            IPAddress address = System.Net.IPAddress.Parse(ipAddress);

            // Create a buffer of 32 bytes of data to be transmitted.
            string data = "aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa";
            byte[] buffer = System.Text.Encoding.ASCII.GetBytes(data);

            // Wait 10 seconds for a reply.
            System.Net.NetworkInformation.PingReply reply = pingSender.Send(address, timeout, buffer);
`
            if (reply.Status == System.Net.NetworkInformation.IPStatus.Success)
                return true;
            else
                return false;
        }`

Padanian, 2022-12-06:
Good morning. I never experienced this bug before, but to be completely honest `timeout` in `Send(address, timeout, buffer);` in `System.Net.NetworkInformation` namespace is of type `TimeSpan` which is a struct and not `int`, both valuetype.
Is this issue related to boxing and unboxing timeout?