rossmann-engineering/EasyModbusTCP.NET #90: Read weird result from InputRegisters when open multiple modbus client

HanochZhu, 2022-11-29:
Hi,
I encounter a weird problem when multiple modbus client running on the same computer. The first client read wrong number from server when the second client launched. For example, The first client read from InputRegister with 1 block.  It shows  correctly  what display on server. After the second client launched, The frist client will read a wrong number, It always looks like -29221 or 9435 such a weird number.

How to solve this problem? Is there any configuration with wrong input? Thanks?

Padanian, 2022-11-29:
You have to consider latency on the server side. Not all servers are so well written to withstand concurrent communication with reasonable latency.

HanochZhu, 2022-11-30:
Thanks! 
I will check it with our server developer.