rossmann-engineering/EasyModbusTCP.NET #116: Is this a legit library

Zapnologica, 2024-10-03:
Good day,

I just wanted to ask if this is a legit and relavent library to use?

Everything is in one class???
Alomst 3000 lines of code in this one file?
No Interfaces or any software design patterns used?

gbellini79, 2024-10-03:
It depends: do you need something that just works or a library that is also beautiful, easily extensible and thoroughly documented?
We fall in the former category and have used it in several project without issues.

Padanian, 2024-10-03:
A single file is much better to mantain that a plethora of useless overhead files. And as @gbellini79 said, it works a charme.
Other than that, what does your message has anything to do with anything?

Zapnologica, 2024-10-03:
So I am not here to throw stones at anyone, or get into a one file vs many file discussion.  It was a genuine question to see if this library is still being supported and maintained and improved.

I come from a software engineering background where file structures, code re-useability, extendability, testability etc are all very important factors to consider before bringing in a library which will need to be maintained and used by many other developers who also follow those coding standards.

I would argue that 3000 lines of code in one file is the complete opposite of easy to maintain.  But let's not get into that.

I assume this code has been ported or orginated from a C or embedded background?

I will test its functionality out as you guys say it works well, and then see if we can maybe adapt it to use some of our coding standards..

Thanks for you replies

closed 2024-10-03 by Zapnologica